ORB-HD/rbdl-orb #14: Do we really need Simple Math

ju6ge, 2019-07-09:
While searching for all the abort statements in RBDL i found some of them in the simple math part of the library. Since Simple Math is "separate" from RBDL I don't just want to add RBDL based error classes to it. And adding it's own error handling seems like overkill. Also I don't really see a use case for Simple Math, since Eigen is well supported! And since the pull request for the ldc fix, mentions issues with Simple Math, the question becomes if it is at all useful to keep Simple Math.

mjhmilla, 2019-08-20:
I have no problems kicking SimpleMath out. Since this thread has been rather quiet, I'll get in touch with Martin to see if he knows of anyone using it other than a quick-to-compile math library for development.

mjhmilla, 2019-08-21:
Martin got in touch with me. Thus far SimpleMath serves these purposes only: to facilitate quick compilation during development and to act as a light-weight math library for MeshUp and Puppeteer. I'm currently waiting to hear from him if there are any actual users who depend on it.

Since you're already working on Meshup and Puppeteer I think you are free to kick out SimpleMath. When you go ahead with that I'd be happy to update RBDL's CMake files to remove it, and to get rid of all of the ifdef's in the source.

mjhmilla, 2019-08-21:
I just got off the phone with Martin. He mentioned two positive points for SimpleMath:

1. Its nice to be able to use RBDL without any dependencies. 
2. It is nice to have everything under once licence. Eigen uses a Mozilla license, which is quite permissive for commercial activity, but is different than the zlib license.

He noted that he's made a new header-only SimpleMath that appears in the dev branch of 
  https://bitbucket.org/rbdl/rbdl/src/default/

Items 1 and 2 are not really strong arguments for keeping SimpleMath in rbdl-orb. Perhaps item 1 has some relevance for teaching as it will make the install process easier on students. I think you are free to make an informed decision about keeping SimpleMath or not: I don't see any big road blocks to either keeping it or removing it. The choice is yours to make.

ju6ge, 2019-08-21:
Okay, thanks for contacting Martin and giving me green light to make a decision. 

In my eyes SimpleMath is a dependency just as Eigen is one and RBDL does use more dependencies than Eigen especially when you count the Addons as well. The difference here is that the SimpleMath code is in our code base and has to be maintained by us. A bugfix in SimpleMath upstream has to be added to our SimpleMath code by hand. It would be much nicer if SimpleMath where a library just like Eigen that can be updated via the system.

I also tested the compile time difference, and to be fair SimpleMath does compile about 5 times quicker. That being said, RBDL with Eigen takes about 2min to compile on my workstation, which is not a long time. So this only is an advantage if you need to compile RBDL over and over again, so a user of RBDL does not need it (especially if you consider the performance difference). And as a developer I think it is more important to have consistent behavior between the version you are using to develop, and the version people than will be using.

I see removing SimpleMath as a simplification of the code base, which I still think is a good idea. Also Meshup and Puppeteer will not be affected by this at all since they have their own copy of SimpleMath in their src tree. And removing it from them is way to much work. 

So I will start working on a pull request that will remove the SimpleMath dependency from RBDL.

closed 2019-09-24 by mjhmilla