ORB-HD/rbdl-orb #17: LuaModelRead ... architecture does not generalize

mjhmilla, 2019-09-24:
At the moment there are two types of functions to read in a lua model:

-LuaModelReadFromFile(filename, model, verbose)
-LuaModelReadFromFileWithConstraints(filename,model,constraint_sets,constraint_set_names,verbose)

Both of these functions read in the multibody model (stored in model) and the second one reads in the model and populates a constraint set. This works fine for now: there is only 1 extra thing you might want to read in. There is a lot more in the pipeline: muscle-torque-generators, line-type-muscles, compliant-contact elements, etc. I would much prefer to have a different architecture than we currently have so that we can easily expand the lua model functions as RBDL grows. For example:

-loadMultibodyModel(filename, modelEmpty,verbose)
-loadConstraintSets(filename, modelLoaded, constraintSets, constraintSetNames)
-loadMuscleTorqueGenerators(filename, modelLoaded, mtgSets)
  ... etc.
-loadCompliantContactElements(filename,modelLoaded,complaintContactElements)

The 'modelLoaded' argument is needed as the other elements (constraint sets) have to refer to the parts of the multibody model. By including the multibody model we can do error checking. While this architecture is quite flexible (useful for the future) it does require that the user is aware of what is in their model and to call the appropriate functions. I'm quite ok with this: RBDL is primarily designed for people with experience programming and with multibody dynamics.

ju6ge, 2019-09-24:
Splitting the loading into different parts sounds like a reasonable idea. I will take a look, but I think this is a feature for the next release.

zapplelove, 2020-07-30:
any plan to support reading from SDFormat?

ju6ge, 2020-07-30:
@zapplelove supporting new formats is currently not planned. But rbdl supports Luamodels and URDF Models so if there is a converter for SDFormat to one of these that could solve your problem. 

You can also implement support for SDFormat yourself, we are happy to accept pull requests. If you want to do this take a look at the urdf- or luamodel-addon which implements the model loading for those formats and create a new addon for SDFormat.

And just as a side note this question would have been much better asked in a new issue instead of posting in an unrelated issue thread.