ORB-HD/rbdl-orb #18: Point contact of wheel with the ground

padhupradheep, 2019-11-11:
Hey!

I'm trying to model a wheeled mobile robot using RBDL. I'm using the floating joint to model the mobile robot(similar to the humanoid robot). Maybe also I will try to use 3DoF joint, later. My question now mainly focuses on how to model the wheel and the ground contact? Should I just use the contact constraint and define the point where it should be touching the surface? Or is there anyother trick?

Thanks!

mjhmilla, 2019-11-11:
Hello,

There are three responses that I can give right now:

1. I have implemented a 2-Dof wheel-plane constraint (and also an ellipse-plane constraint) for RBDL, but it is not yet in the code base. This is on my list of to-do's. In case I can actually help in this regard: by what date do you need the wheel-plane constraint?

2. If you know how to derive the constraint equations and you are familiar with C++ you can implement this pretty quickly on your local copy of RBDL. We've implemented a generic interface class for constraints (include/rbdl/Constraint.h) that you need to extend for your own constraint. To guide your implementation have a look at src/rbdl/Constraint_Loop.h, src/Constraint_Loop.cc, and the test code (rbdl-orb/tests/LoopConstraintsTests.cc) to help you get started. 

3. You can also implement your wheel-ground movement as a CustomJoint. I'd say at this point you have to have a fair amount of experience making multibody models to not make an error here. Assuming that this will work for you, you can see some examples of using the CustomJoint interface by looking at the code in rbdl-orb/tests/CustomJointTests.cc. Again you'll need to have some familiarity with C++ and be familiar with multibody dynamics to pursue this option.

Cheers,

Matt

padhupradheep, 2019-11-11:
Hey,
Thanks for the swift response! 

1. It's totally up to you. For me It is to be used in my master's thesis work, so I think maybe within a month. 

2. and 3. I have some knowledge from Roy Featherstone book. Maybe, I can also give it a try from the pointers which you have given me. I would mostly go for custom constraints, rather than going for the custom joint.

 That being said, I will post any further queries, if I have during the development.

Thanks alot once again! 

Pradheep

padhupradheep, 2019-11-14:
Hey, I have a small question. 

For your 3 point of having a custom joint, why can't we just have a virtual planar joint between the wheel and the surface as shown below ? 

`    Planar_1 = Joint (
     Math::SpatialVector (0., 0., 0., 1.,  0., 0.),
     Math::SpatialVector (0., 0., 0., 0., 1., 0.),
     Math::SpatialVector (0., 0., 1., 0., 0., 0.)
     );`

Thanks for your time!

mjhmilla, 2019-11-18:
The joint that is described above corresponds to a planar joint. This will not be the same as a wheel-plane joint: the wheel will spin when its axis translates. This joint misses that coupling.

ju6ge, 2020-09-08:
Closing this issue, because it has seen no activity and only contains questions and no actionable problems.

closed 2020-09-08 by ju6ge