ORB-HD/rbdl-orb #31: Closed-loop inverse dynamics support

FaridTavakol, 2020-07-30:
I am trying to calculate the inverse dynamics for a closed-loop system. I was wondering if the library supports such a method?

zapplelove, 2020-07-30:
I want this too. And in repo rbdl,  the author said it supports closed loop, but you should write a lue file to describe the system. but I want to use sdf file

ju6ge, 2020-07-30:
@FaridTavakol If you want to model a closed loop system i suggest you take a look at constraints, there are loop constraints with which you can attach two segments and create closed loops. Take a look at the documentation for more information.

FaridTavakol, 2020-07-30:
> @FaridTavakol If you want to model a closed loop system i suggest you take a look at constraints, there are loop constraints with which you can attach two segments and create closed loops. Take a look at the documentation for more information.

Thank you @ju6ge for the reply! I have extensively gone through the documentation (especially the examples for the four-bar model as well as the 3D slider) and modeled several systems using RBDL. I have been using AddLoopConstraint and binding it to my model for creating my robot model. However, there seems to be a method for the Forward dynamics called "ForwardDynamicsConstraintsDirect" that is used for the closed-loop systems. But there seems to be no such a method for the InverseDynamics, and when I use just the InverseDynamics for gravity compensation in my model the output torque does not seem to work.

ju6ge, 2020-07-30:
@FaridTavakol to be honest that particular problem is a bit over my head since I am not yet to deep into the mathematics involved. Maybe @mjhmilla knows more about this since he has done a lot of work with constraints already.

mjhmilla, 2020-08-10:
Dear @FaridTavakol and @zapplelove 

There are two specialized methods in RBDL that allow you perform an inverse dynamics analysis on a multibody system with constraints. To get started:

1. Read the doxygen for InverseDynamicsConstraints & InverseDynamicsConstraintsRelaxed
2. Next read about the function SetActuationMap, as you will need to use it prior to calling either InverseDynamicsConstraints or InverseDynamicsConstraintsRelaxed.
3. If you are unsure if your system is fully actuated there is a function provided that can answer this function can analyze the system to tell you: isConstrainedSystemFullyActuated.
4. For usage examples of these functions please see: tests/InverseDynamicsWithConstraintsTests.cc
5. Finally, if you are using InverseDynamicsConstraintsRelaxed as part of an optimal control simulation, you might want to use a branch of RBDL that I've worked on https://github.com/mjhmilla/rbdl-orb/tree/invDynConRelaxUpd. From my simulations the weighting matrix that is used in InverseDynamicsConstraintsRelaxed is much more amenable for use in an optimal control problem than the default version.

Best of luck,

Matt

FaridTavakol, 2020-08-12:
Thank you very much for your reply! I was able to run the inverse kinematics for a simple closed-loop system and it seemed to work fine.

closed 2020-08-12 by FaridTavakol