ORB-HD/rbdl-orb #35: Python bindings COM values are all 0

fjandrad, 2021-02-05:
I run a small python script just to check the parameters from a urdf model. I saw that the values of the COM of the links are all 0. While the values in the urdf file are not. 
## python

Python code
```py
    for body in model.mBodies:
        print(body.mMass)
        print(body.mCenterOfMass)
        totalMass += body.mMass
    print(totalMass)
```
Output: 
```
12.3321
[ 0.  0.  0.]
1.02901
[ 0.  0.  0.]
0.69621
[ 0.  0.  0.]
3.97852
[ 0.  0.  0.]
2.86055
[ 0.  0.  0.]
0.68213
[ 0.  0.  0.]
.
.
.
0.015
[ 0.  0.  0.]
0.05
[ 0.  0.  0.]
1.11967
[ 0.  0.  0.]
1.86234
[ 0.  0.  0.]
68.62625000000004
```

## c++ version
Result of c++ version of the same code:
```c++
double totalmass=0;
  for( int i=0;i<=model->q_size ; i++   ){
      std::cout <<i<<"-th Mass:" << model->mBodies[i].mMass<< std::endl;
      totalmass+=model->mBodies[i].mMass;
      std::cout << "COM:"<< model->mBodies[i].mCenterOfMass<< std::endl;
  }
  std::cout <<totalmass;
```

```shell
0-th Mass:12.3321
COM: -0.0120802
-0.00441642
  0.0214326
1-th Mass:1.02901
COM:-0.02571
       0
 0.02557
2-th Mass:0.69621
COM: 0.00057
 0.00881
-0.01125
3-th Mass:3.97852
COM: 0.14784
-0.01646
 0.01793
4-th Mass:2.86055
COM:0.14618
-0.0022
0.02189
5-th Mass:0.68213
COM:-0.00028
 0.01119
 0.00859
6-th Mass:2.01043
COM:  0.0611293
 0.00336163
-0.00492278
.
.
.
100-th Mass:0.05
COM:0
0
0
101-th Mass:1.11967
COM: 0.01819
 0.00248
-0.00517
102-th Mass:1.86234
COM: 0.0205854
 -0.082619
0.00149924
68.6263
```

## Conclusion
There is probably something lost in the bindings.

Florrr, 2021-02-18:
Hey!
we had exactly the same problem a few weeks ago. It seems that `model.mBodies` does not point to the correct objects. We solved the issue by using the provided functions `model.GetBody(id)` instead.
e.g.:
```
for i in range(0,model.mBodies):
    model.GetBody(i).mCenterOfMass
    model.GetBody(i).mMass
```

fjandrad, 2021-02-26:
> Hey!
> we had exactly the same problem a few weeks ago. It seems that `model.mBodies` does not point to the correct objects. We solved this issue by using `model.GetBody(id)` instead.
> e.g.:
> 
> ```
> for i in range(0,model.mBodies):
>     print(model.GetBody(i).mCenterOfMass)
>     print(model.GetBody(i).mMass)
>     totalMass += model.GetBody(i).mMass
> print(totalMass)
> ```

Thanks. Yeah I guess I can work with that.

PR #38 (2021-03-25): Python wrapper
Solves the issues regarding the python binding of vector class elements to the corresponding python class elements. Their content is now also accessible in python. I also added a small test to "test_wrapper.py" and fixed the tests and property "mJointType" of class "Joint" with respect to the new joint type handling introduced in pull request #29. 
Note that many of the changes are just tab-to-spaces conversions. All changes in the files "crbdl.pxd", "rbdl_loadmodel.cc " and  "rbdl_ptr_functions.h " are  tab-to-spaces conversions.

Closes #35
Closes #36

FYI: The Python test cas [... cut]
closed 2021-03-25 by ju6ge