ORB-HD/rbdl-orb #47: Method of changing contact constraints after binding?

makeller1, 2021-04-26:
Bind() requires ConstraintSet to remain unchanged after binding. So how can we change contact constraints that are frequently enabled/disabled (e.g. walking robots). Is the best practice to clear the constraints, add the new set of constraints, then bind again?

mjhmilla, 2021-04-27:
The easiest way to use multiple constraint sets in RBDL is to define all of the different ConstraintSets and then switch between them. We frequently do this when using RBDL models as part of a multi-phase optimal control problem. You must take care to formulate the switching conditions correctly (e.g. at contact applying impulses, and when breaking contact when normal forces go to zero) for this to work. 

To put multiple constraints in the Lua model you just simply use this syntax:

constraint_sets = {
	 constraint_set_one = {
	...
	},
	constraint_set_two = {	
	...
	},
...
}

Then you can use the function LuaModelReadFromFileWithConstraints from the luadmodel addon to load the model and the vector of constraint sets. For an example, please see:

Millard M, Sreenivasa M, Mombaur K. Predicting the motions and forces of wearable robotic systems using optimal control. Frontiers in Robotics and AI. 2017 Aug 30;4:41.

This approach is obviously only appropriate for systems in which you can specify and define the full list of possible contact pairs. For problems in which is this not possible (due to an overwhelming number of contact pairs) things will be a bit more challenging for you. One option would be to use a compliant contact model (see the in the bouncing ball example in the 'examples' folder). Though I have plans on expanding the contact addon in RBDL I haven't gotten to this yet. There are other options, though these require considerable amounts of work to implement.

makeller1, 2021-04-29:
Thank you for the quick response!  

The specific problem is for a quadruped, so the possible contacts are known. I am reading from a URDF, not a Lua model, so I am adding the constraints to a ConstraintSet with AddContactConstraint(). 

For instance, for four feet in contact with the ground, I add 4 contact constraints with the function above, and then bind that to the model. However, when the quadruped only has 3 feet in contact, I am not sure how to change the constraints when calling a function like CalcConstraintsJacobian(). If I try to bind another ConstraintSet to the model, an error is thrown.

ju6ge, 2021-04-29:
Hey, i think you might be miss interpreting the error message. 

What the Bind() functions does is initialize the memory required to do the computations you want. It needs to know which model it is applied to, to be able to do this correctly, but the model does not have any value that saves which constraint set is bound. So the function name Bind() is a bit misleading. The Bind() function only needs to get called once after the constrain set has been created and all constrains have been added. Which constraint set then gets used is just a matter of which constraint set you pass to the function you are calling.

An example for using multiple constraint sets can be found in the test code:
- defining multiple constraint sets https://github.com/ORB-HD/rbdl-orb/blob/master/tests/Human36Fixture.h 
- using multiple different constraint sets https://github.com/ORB-HD/rbdl-orb/blob/master/tests/MultiDofTests.cc (from line 816 onward)