ORB-HD/rbdl-orb #49: Merging for CasADi handling

pariterre, 2021-05-06:
Hello!

I've worked on a fork of rbdl last year and ported the code so it can either use Eigen, SimpleMath or CasADi as backend (https://github.com/pyomeca/rbdl-casadi). Since the version I am hosting is not evolving while yours is, I was wondering if you would be interested in merging what I have done to your fork. 
Most of the code remains the same except for some algorithms (all iterative algorithms, actually) which has to be ignored using CasADi (by using macro guards). I did not have a look at your fork, so I am not sure how much work it would take though.

Moreover, if we do that, I could make the rbdl on conda-forge to point at this repository instead of mine. Like this you could improve your reach be providing a precompiled version.

ju6ge, 2021-05-08:
Hey, 

thanks for reaching out. The entire point of this fork of rbdl was that development on the orignal version has stalled for pretty long, while the research projects are moving on to the next tasks. So any new features are welcome :)
That being said we are only a small team and try to do our best with the time we can afford to put into this.

To merge your casadi version with our rbdl version we probably need to address the big differences that we have with the original rbdl:
- we have added error handling using std:exceptions because the original implementation was not compatible with using rbdl in other applications such as [rbdl-toolkit](https://github.com/orb-hd/rbdl-toolkit)
- the constraint system was completely redesigned to make use of polymorphism to make developing new constraints more accessible and less error prone
- we actually dropped the simple math library to reduces complexity, its original purpose was to speed up compile time which these days is not that long anymore because computers are just way faster. Also SimpleMath does not implement all functions required to run the muscle addon so the test suite would have been incomplete with SimpleMath.

If I have some time next week I can take a closer look at your fork and try to determine how much work this would be.

Kind regards,
ju6ge

pariterre, 2021-05-08:
Allrighty,

Let me know. I plan to be in Germany in about a month and a half. If it turns out to be interesting, or if you want us to walk through together the changes I made, we could organize something at that moment (either virtual or in person, depending on the pandemic situation at that moment)

pariterre, 2021-05-08:
I just thought, Katja is no longer in Germany, and Martin is probably no more with her anyway! Where are you guys based?

ju6ge, 2021-05-08:
Well we (ORB) is still based in Heidelberg, there are still some projects active that have yet to be completed, while the new professor replacing Katja is still to be determined.

ju6ge, 2021-05-08:
Due to Covid most of us are working from home anyway. But I think a meetup should still be possible ;)

pariterre, 2021-05-09:
Excellent! I'll be in Germany around the end of June and beginning of July (or anytime on Zoom :P )

ju6ge, 2021-05-12:
Hey pariterre,

so i took a look at you fork to determine how we could merge our rbdl versions. I think this would be quite straight forward.

1. Since we removed SimpleMath from our version the perprocessor branches selecting which math backend to use should be simpler, but there is more code that needs to be looked through for places where that needs to happen
2. This also means that we would only need a single CMakeOption to select if you want to use casadi or not
3. It does not seem like the test code was updated, but nevertheless some simple tests using casadi capabilities would be nice and would also serve as some example code (I am not familiar with casadi so I don't know how that would look)
4. Add some information to the README/Documentation about casadi, so that people who are unsure what to use get some information on the advantages and drawbacks of casadi as opposed to eigen.

Do you have any more thoughts? 

Apart from that I would also be interested to learn about casadi and how you are using it with RBDL. I probably have some time next week to do an online meeting ;)

Kind regards,
ju6ge

mjhmilla, 2021-05-13:
If I'm not mistaken, it's good to hear from you Benjamin!

It would be good to see you as well, if you're in the neighbourhood. At the moment it's mostly Felix and I who are maintaining/adding to RBDL. As long as you schedule your visit ahead of time, I expect that we will both be able to see you in person.

I took a quick look at rbdl-casadi and the update seems to be pretty small, unless I was looking in the wrong place. Is the update really just replacing Eigen with CasADi? I have a passing familarity with CasADi: what in your mind is the biggest benefit of using CasADi instead of Eigen? How do you use it in your own work?

With kind regards,

Matt

pariterre, 2021-05-16:
Eh @mjhmilla !
It is indeed me haha! Good to hear from you too! That would be so nice to see you both!

I will not be available for a zoom call before the beginning of June. I just arrived yesterday in Denmark and will focus on the experiment here for the next two weeks, alongside with the CSB next week. From that point, I am available for a virtual meeting. 
I am going to Germany around the 11th of June and stay there for a full month. I'll be based in Stuttgart but I can take the bus or train to Heidelberg (or elsewhere) if this ever become possible to visit you in person

@ju6ge I can definitely give you a quick tour of the Casadi capabilities. In brief, when using Casadi, you are constructing a graph of the code when calling a function instead of actually evaluating the code. Thereafter, you can derive the graph itself to get the derivative of the underlying algorithm. Using this technique (called algorithmic differentiation) you can get the derivative of any code, assuming there is no branching (i.e., if statements), or only very small ones. The precision and speed to evaluate the derivative (after the the code graph is constructed) is mostly the same as an analytic equation, even sometime faster. In bioptim, we are using biorbd, which is a software on top of RBDL (with RBDL compiled using Casadi variables) to do optimal control. Thanks to Casadi, it is trivial to get the derivative of the forward dynamics even with contact (which is not trivial at all otherwise). 

The main downside is that iterative algorithms can't be used. The reason is that the ending condition are usually a threshold. However, since there no if statements*, it is not possible to have an exit condition. Therefore, in RBDL, there are entire functions that are ignored when compiling using casadi.
* there is actually an if_else statement, but it is designed for one liner if)

@ju6ge, here are my quick answers to your questions:

> Since we removed SimpleMath from our version the perprocessor branches selecting which math backend to use should be simpler, but there is more code that needs to be looked through for places where that needs to happen

Okay, we can have a look at this. But we somehow need a selector like the one RBDL had when selecting between eigen and simple. The reason is that eigen and Casadi cannot work alongside, it is one or the other. 

> This also means that we would only need a single CMakeOption to select if you want to use Casadi or not

This is indeed what I am doing right now in biorbd. Again, please note that by definition, Eigen and Casadi are incompatible. Actually, Casadi and anything-else-that-defines-its-own-variables are incompatible by definition, unless the variables are converted to Casadi variables (which may or may not be trivial).

> It does not seem like the test code was updated, but nevertheless some simple tests using casadi capabilities would be nice and would also serve as some example code (I am not familiar with casadi so I don't know how that would look)

Testing Casadi is a pain in the neck. The reason is that Casadi does not use numerical variables. When you write a function, it constructs the graph of the code. You have to convert this graph into a Casadi function and then you can evaluate it. In the biorbd suite, I actually do that for testing. In order to reuse the same code in Casadi and eigen, I created a bunch of preprocessor macro that create these Casadi functions when needed. This is something that we should discuss verbally though...

> I am not familiar with Casadi so I don't know how that would look
Unless you need the derivative of the code, there is no way you would ever use Casadi as a backend. This is quite a mess... I am totally for giving example codes, but someone who uses the Casadi backend must know how to use casadi in the first place. It is not something you just turn on and works. It drastically changes the behavior of the code and how you use code, and should therefore be used by those who know how and specifically need the derivative of the code. Again, this does not prevent from providing examples, but bear in mind that it will always look cryptic for someone who does not know Casadi beforehand. 

> Add some information to the README/Documentation about casadi, so that people who are unsure what to use get some information on the advantages and drawbacks of casadi as opposed to eigen.

Definitely! On the top of my head:

The main advantage (but worth all the disadvantages) is 
- To have the derivative (and most of the time the second derivative) of any (Casadi compatible) function and expression for free. 

The main disadvantages are
- It is not possible to use if statements (a casadi if_else exists, but duplicates the graph and therefore uses A LOT of RAM, and is designed for one liner if statements). 
- Because if statement cannot be used, so while loops. Some iterative based algorithms are therefore not compatible (in other words, the code has to tell the full story and should not depends on the state of the variables)
- The functions or expression cannot be used directly to evaluate them anymore. They have to be converted into Casadi functions and then evaluated, which is tedious and can be slow (Casadi itself is fast, but creating graphs is slow and should be done only at the very end of the process, that is when numerical values are needed). This also makes debugging completely insane.

I think that is it! Haha sorry for the long message! See you soon hopefully!

mjhmilla, 2021-05-28:
Just a quick note: @pariterre I live in Stuttgart. Please contact me when you get here.

closed 2022-01-28 by ju6ge