ORB-HD/rbdl-orb #50: Documentation

makeller1, 2021-05-07:
First of all, thank you all for still maintaining the RBDL project. The learning curve is steep for this library, but it's still a very strong candidate, especially since it's so lean. I think improving the documentation would make it even more competitive. 

As a suggestion for a simple way to start, could we open a discussion page for this repo that we could all add helpful tips? For instance, the function Join() in Body has the important note: "Both bodies have to have their inertial parameters expressed in the same orientation.". But, a user would never see this documentation if they just use the API addBody. Another example, the calcPointJacobian6D returns the Jacobian in the base frame, but this is stated no where in the docs. 

An even better step would be to host ORB's RBDL documentation similar to how the original currently hosts their docs. But, I think a lot of us would benefit from even just the discussion forum.

ju6ge, 2021-05-08:
Hey, 

you make great points. More documentation is very much needed, as well as open discussion about the troubles people are facing. But not just on RBDL but also the tools used with RBDL in conjunction and because the API documentation of RBDL is quite complete the focus has been lately more on the tools, like how to create lua models [RBDL-Toolkit-Wiki](https://github.com/ORB-HD/rbdl-toolkit/wiki/Creating-Models).

Now the documentation of RBDL also needs some love and setting up a Github Pages Documentation page is definitively on my Todo-List. If you have improvements regrading specific documentation points I would suggest you send us a Pull Request. We will happily merge them ;)

Regarding more discussion, I think it would be best to do that in the issues of this repository. That way we don't have to setup a different forum that people would need to check regularly in order to notice new questions and answer them. Opening an issue here will result in more people actively working on RBDL to see it and be able to address it. 

Kind regards,
ju6ge

ju6ge, 2021-05-18:
There is now a github pages documentation page ;) https://orb-hd.github.io/rbdl-orb-doc/index.html

makeller1, 2021-05-18:
That’s great news! Thank you!

closed 2022-03-26 by ju6ge