ORB-HD/rbdl-orb #52: URDF reader regression

PeterQLee, 2021-05-22:
Hello,

I noticed that there seems to have been a regression in the urdf reader within the past several months. 
Consider the commit 194bd0537f5452d621713cb045c44b2324f16be9 from September 9, 2020. If one runs `rbdl_urdfreader_util -c test_urdf.urdf` (see attached file [test_urdf.txt](https://github.com/ORB-HD/rbdl-orb/files/6526894/test_urdf.txt)) you get

```
Model loading successful!
arm_right_1_link:             -0.00011   -0.18627    0.00701
arm_right_2_link:            -0.226629  0.0499686   -0.06294
   Model COM:             -0.217618 -0.00187782  -0.0292106
  Model mass: 3.14007
```

Now consider doing the same thing with the most recent commit on master (0122ed84c4f16d53be7bafcca9587f28cdf606ea).

```
Model loading successful!
arm_right_1_link:                 -nan       -nan       -nan
arm_right_2_link:                 -nan       -nan       -nan
   Model COM:                 -nan       -nan       -nan
  Model mass: 0
```

It looks like something broke the mass/inertia parsing sometime since Sept. 9. I notice that [rbdl_urdfreader_util.cc](https://github.com/ORB-HD/rbdl-orb/blob/master/addons/urdfreader/rbdl_urdfreader_util.cc) has had some significant changes since 194bd0537f5452d621713cb045c44b2324f16be9 in terms of data structure layout. I'm not sure if this is the issue or if has to do with a change in https://github.com/orb-hd/URDF_Parser .

ju6ge, 2021-05-23:
Hey,

thank you very much for reporting this issue. I have tracked down the bug, it actually was a fairly stupid copy and paste bug, that lead to the incorrect inertial data being set on every link except the root link :see_no_evil: 

I fixed this with PR #53, and will merge it as soon as all of our checks have passed.

Kind regards,
ju6ge

ju6ge, 2021-05-23:
#53 merged

closed 2021-05-23 by ju6ge

PeterQLee, 2021-05-23:
Awesome! thanks for looking into it