ORB-HD/rbdl-orb #57: Python build errors in a VM

nag92, 2021-06-21:
I am trying to build RBDL with the Python binding in a VM and I am getting the following error.
I think that it has something to do with Cython but not sure. What could be the cause of this issue? 

```
In file included from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1809:0,
                 from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                 from /usr/lib/python3/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from /home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:505:
/usr/lib/python3/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx: In function ‘void __Pyx__ExceptionSave(PyThreadState*, PyObject**, PyObject**, PyObject**)’:
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82107:21: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_type’; did you mean ‘curexc_type’?
     *type = tstate->exc_type;
                     ^~~~~~~~
                     curexc_type
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82108:22: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_value’; did you mean ‘curexc_value’?
     *value = tstate->exc_value;
                      ^~~~~~~~~
                      curexc_value
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82109:19: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
     *tb = tstate->exc_traceback;
                   ^~~~~~~~~~~~~
                   curexc_traceback
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx: In function ‘void __Pyx__ExceptionReset(PyThreadState*, PyObject*, PyObject*, PyObject*)’:
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82116:24: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_type’; did you mean ‘curexc_type’?
     tmp_type = tstate->exc_type;
                        ^~~~~~~~
                        curexc_type
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82117:25: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_value’; did you mean ‘curexc_value’?
     tmp_value = tstate->exc_value;
                         ^~~~~~~~~
                         curexc_value
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82118:22: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
     tmp_tb = tstate->exc_traceback;
                      ^~~~~~~~~~~~~
                      curexc_traceback
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82119:13: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_type’; did you mean ‘curexc_type’?
     tstate->exc_type = type;
             ^~~~~~~~
             curexc_type
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82120:13: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_value’; did you mean ‘curexc_value’?
     tstate->exc_value = value;
             ^~~~~~~~~
             curexc_value
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82121:13: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
     tstate->exc_traceback = tb;
             ^~~~~~~~~~~~~
             curexc_traceback
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx: In function ‘int __Pyx__GetException(PyThreadState*, PyObject**, PyObject**, PyObject**)’:
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82176:24: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_type’; did you mean ‘curexc_type’?
     tmp_type = tstate->exc_type;
                        ^~~~~~~~
                        curexc_type
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82177:25: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_value’; did you mean ‘curexc_value’?
     tmp_value = tstate->exc_value;
                         ^~~~~~~~~
                         curexc_value
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82178:22: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
     tmp_tb = tstate->exc_traceback;
                      ^~~~~~~~~~~~~
                      curexc_traceback
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82179:13: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_type’; did you mean ‘curexc_type’?
     tstate->exc_type = local_type;
             ^~~~~~~~
             curexc_type
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82180:13: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_value’; did you mean ‘curexc_value’?
     tstate->exc_value = local_value;
             ^~~~~~~~~
             curexc_value
/home/user/rbdl-orb/rbdl-build/python/rbdl-python.cxx:82181:13: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
     tstate->exc_traceback = local_tb;
             ^~~~~~~~~~~~~
             curexc_traceback
python/CMakeFiles/rbdl-python.dir/build.make:68: recipe for target 'python/CMakeFiles/rbdl-python.dir/rbdl-python.cxx.o' failed
make[2]: *** [python/CMakeFiles/rbdl-python.dir/rbdl-python.cxx.o] Error 1
CMakeFiles/Makefile2:122: recipe for target 'python/CMakeFiles/rbdl-python.dir/all' failed
make[1]: *** [python/CMakeFiles/rbdl-python.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

```

ju6ge, 2021-06-21:
I have build rbdl on multiple different VM's and have not seen this issue. That is to say the information that you are running in a VM does not tell me anything about the system. Please provide more information: What OS is the VM running. What compiler are you using and what version of dependencies have you installed? Specifically what cython version are you using.

My guess is that this probably is an Issue with the cython installation, because the rbdl-python.cxx is not a source file, but gets generated with cython during the build process.

nag92, 2021-06-21:
Sorry, that I forget to include that information. Is this the information you need? Please let me know if you need anything else. 

0S: 18.04
Python verison: 3.6.9
cython: 0.26.1

ju6ge, 2021-06-21:
Your cython version is outdated … please try using cython version 0.29.23 it is the version you get when you install cython via pip:
`pip3 install cython`.

I think the version you are using is shipped with ubuntu 18.04 package mangement. To avoid conflicts you should first remove that version.

nag92, 2021-06-21:
thank you so much that worked perfectly!!

closed 2021-06-21 by nag92