ORB-HD/rbdl-orb #61: Inconsistent Dynamics handling of UpdateKinematics

ju6ge, 2021-07-12:
This issue was first discussed via internal mails:

@pariterre :
This was not a mistake of mine finally. I found the actual bug (actually how to trigger it) :
Calling the ForwardDynamicsConstraintsDirect twice in a row does not return the same answer in the with external_force if the user does not call update_kinematics prior to the first call.
That means that the update_kinematics is probably called somewhere in the function, but not before it is actually necessary.

@mjhmilla :
if I understand correctly, this is not a bug. By design, Martin has put the responsibility on the user to call UpdateKinematics before calling any of the dynamics functions. This means a careful user can get higher performance out of RBDL by calling UpdateKinematics once prior to evaluating a number of dynamics functions in which the state is fixed. This means that if you call ForwardDynamicsConstraintsDirect (or any other function that depends on the model's kinematics) you must calling UpdateKinematics once prior to the first function call.

@pariterre :
This is indeed the case for most of the functions, but usually they have the "update_kin" flag that can be passed. For the ForwardDynamics, this is not the case. The current behavior of all forward dynamics functions (except for the ForwardDynamicsConstraintsDirect when passing an external force) is to update the kinematic for the user. This can be seen by the fact that even for ForwardDynamicsConstraintsDirect if you call the function twice, the kinematics is actually updated. That means there is a call to update_kinematics which is automatically performed by the ForwardDynamicsConstraintsDirect. It is however not done at the very beginning of the function and is therefore returning a false answer. In the other case, if the user updates the kinematics by themselves, that means two calls to the update kinematics are performed (one by the user and one by the function itself), which is by design a loss in performance.

So either way, there is a design flaw. Either the user should be the one responsible for the call to the update_kinematics (and therefore should not be done a second time in the function) OR the function should do it for the user but at the right place in the code (as for all the other forward dynamics functions).

pariterre, 2021-07-12:
I am adding the last message from @mjhmilla 

Hey @pariterre,

I've started looking into this in more detail, and you're right, there is a problem:

-ForwardDynamicsConstraints direct call a number of functions that have the update_kinematics flag, and these functions are all set to false: NonlinearEffects, CompositeRigidBodyAlgorithm, CalcConstraintsJacobian, CalcConstraintsPositionError, CalcConstraintsVelocityError

-On line 881 of Constraints.c the position level transformations of all of the bodies are updated. This has two problems with it:

a. It is always called.
b. It is called AFTER the call to NonlinearEffects and the CompositeRigidBodyAlgorithm

I would propose this fix:

1. Add an update_kinematics flag to the argument list that has a default value of true
2. If update_kinematics is true call:
UpdateKinematicsCustom(model, Q, NULL, NULL)
3. Remove the update to X_base which appears on line 881 currently.

This should work just fine since the velocity level is updated by NonlinearEffects, and the acceleration level is updated by CompositeRigidBodyAlgorithm.

I have two questions remaining:

1. See any problems with that?
2. Does this work:
  a. I just make this fix and issue a pull request.
  b. You guys look at it, and merge it
  c. The initial Casadi merge pulls this in from upstream?

If you'd prefer something different for #2 please let me know.

Cheers,

ju6ge, 2021-07-12:
I would be in favor of making the flag "false" by default, since it is the default every where else.

mjhmilla, 2021-07-13:
Hello @ju6ge @pariterre 

I've made the fix to the problem @pariterre identified and issued a pull request here: 
  https://github.com/ORB-HD/rbdl-orb/pull/64

In contrast to the suggestion of @ju6ge, I've set the new update_kinematics flags to be true by default. Why? A niave user will be guaranteed then to have the function call return the correct answer. An advanced user can read the documentation in detail and optimize their code accordingly.

ju6ge, 2021-07-13:
Great! I see the logic with setting it to default true. 

Looking at the PR almost everything seems fine. There is one python test failing which is related to the updated function. Can you take a look?