ORB-HD/rbdl-orb #62: after successfully installed rbdl,  I got an error when load model

visonpon, 2021-07-13:
Hi, thank you for providing this tools, but  after installed normally, I got an error when load model.

```
terminate called after throwing an install of 'urdf.URDFParseError'
what() Error while parsing link 'left_knee_limb' unknown shape type 'capsule'!
```

ju6ge, 2021-07-13:
Hey, so the shape you are specifying is not a standard shape. The (ROS URDF Wiki)[http://wiki.ros.org/urdf/XML/link] shows the possible values for a link. As you can see "capsule" is not a valid geometry. If you wish to use this shape, for visualization you need to use a mesh geometry and specify a mesh file ;)

As a note for me, it might be nice to have an option to specify not caring about the visualization attributes of the model and ignore them for people only interested in using rbdl for calculations.

ju6ge, 2021-09-15:
Our urdf parser now also supports using capsule as a geometry. So this was fixed with #74.

closed 2021-09-15 by ju6ge