ORB-HD/rbdl-orb #69: error when rbdl.CalcCenterOfMass()

zxiaoyusq, 2021-07-21:
the following code:
`rbdl.CalcCenterOfMass(model, q, qdot, qddot, CoM)`
Gives the following error:
```
File "/home/lmd/dev/physcap_demo_release-master/Utils/core_utils.py", line 86, in get_projected_CoM
    rbdl.CalcCenterOfMass(model, q, qdot, qddot, CoM)
  File "rbdl.pyx", line 3150, in rbdl.CalcCenterOfMass
    if qddot and com_acceleration is not None:
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

```
and the value:
[image]

Could you point where I did wrong? Thank you for your help.

ju6ge, 2021-07-21:
Hey can you post the code snipped otherwise it is hard to tell what exactly is causing the issue.

antoinedelplace, 2021-09-14:
Same as #71.
Should be fixed by #72

ju6ge, 2021-09-14:
closed with #72

closed 2021-09-14 by ju6ge