ORB-HD/rbdl-orb #70: URDF Reader error even after building with RBDL_BUILD_ADDON_URDFREADER=ON

Prathamesh-Saraf, 2021-07-21:
[image]

The required files are generated after building with RBDL_BUILD_ADDON_URDFREADER=ON  but I still get this error when I run the example

ju6ge, 2021-07-21:
You probably need to use the `-I` option to specify the include directory ;)

I strongly recommend using Cmake, then things like this become a lot easier. With the merge of the Casadi PR we will also add FindRBDl.cmake files, so that all you need to do is use find_package(RBDL COMPONENTS URDFREADER) within your CMakeLists.txt.

Prathamesh-Saraf, 2021-07-26:
Thanks for your reply! I tried using -I but no success. Also, I have CMake 3.18.4 version installed. I'm still getting the same error

ju6ge, 2021-07-26:
I suggest you read up on how to use cmake.

All the examples come with their own CMakeLists.txt files so all you need to do is create a build directory. Within the build directory run `cmake` and point it to the CMakeLists.txt file of the example. And afterwards run `make`.

closed 2021-07-26 by ju6ge