ORB-HD/rbdl-orb #73: [Question] Is it possible to upgrade urdfdom version to handle capsule shape ?

antoinedelplace, 2021-09-14:
Is it possible to upgrade urdfdom version to handle capsule shape?

I have tried using ROS melodic urdf parser but still can't handle capsule.

Could you help me please?

ju6ge, 2021-09-14:
If you are using the ROS urdfdom library then it's capabilities need to be able to handel capsules. RBDL does not care about the shape information, it only cares about inertial/mass values (and of course the kinematik structure). Shapes are only "eye candy" that is used when visualizing results. If you are using ROS urdfdom make sure to set the CMAKE flag for that. 

Per default RBDL now uses a different [urdf parser](https://github.com/orb-hd/URDF_Parser), the reason for this is that urdfdom did not have nice error handling and loading urdfmodels could lead to segmentation faults for various reasons, this was not acceptable when using rbdl in downstream applications like [RBDL-Toolkit](https://github.com/orb-hd/rbdl-toolkit). If you absolutely need capsules you could add them to our URDF_Parser library. We can't do anything about what ROS urdfdom supports. This also should not be very hard, take a look at the geometry.h/cpp. Adding some code there to extract the capsule parameters from the xml should be all you need.

ju6ge, 2021-09-15:
Fixed with #74, make sure to also update the submodules after pulling the latest commits ;)

closed 2021-09-15 by ju6ge