ORB-HD/rbdl-orb #78: [Question] How to get joint order in model when parsing URDF ?

antoinedelplace, 2021-10-05:
Hi !

I am having some trouble getting the joint order in the RBDL model corresponding to the **q** vector.

In my URDF file, I have **74 joints** but when loaded with ```rbdl.loadModel```, I have : 
- ```model.q_size = 79```
- ```len(model.mJoints) = 76```
- ```len(model.mBodyNameMap) = 77```

All my joints are revolute joints except one joint which is floating. This specific joint is the only one having 6 DoF.
It seems to me that my floating joint has been split to 6 scalars in **q** (74-1+6=79) but in which order ?

1. In the [RDBL documentation](https://rbdl.github.io/d7/d19/struct_rigid_body_dynamics_1_1_model.html#a1ad0e2e7f7c5f1545934069b71286216), it is said :
> additional values for the w-component of the Quaternion is stored at the end of **q**

2. I tried to look at the code here without finding an answer :
https://github.com/ORB-HD/rbdl-orb/blob/38908412e35b1659737123873b4f8297dc22007e/addons/urdfreader/urdfreader.cc#L192

3. Also, in the python wrapper I am using, it sadly does not seem to have any ```model.mJointsNameMap```

Could you help me on this please ?

ju6ge, 2021-10-05:
When building rbdl with the urdfreader addon it should have created a binary called urdfreader_util. This binary can be used to load and inspect urdf model files with rbdl. It has an option `-d` to generate an degree of freedom overview.

I hope this helps. Kind regards
ju6ge

antoinedelplace, 2021-10-05:
Awesome ! Thank you !! :D

closed 2021-10-05 by antoinedelplace