ORB-HD/rbdl-orb #80: rbdl not installed into /usr/local

zcoco99, 2021-11-09:
Hi, I am new to CMAKE and RBDL. I've been trying to set up RBDL for a while now, following the documentation, but I still couldn't build the pendulum example. The issue is at the CUSTOM_RBDL_PATH setting and I'm wondering if I can get some help.
I have made a rbdl-build directory in /home/username and built the rbdl-orb version using `cmake -D CMAKE_BUILD_TYPE=Release ../rbdl-orb`. 
However when I set CUSTOM_RBDL_PATH to '/usr/local', the RBDL library can't be found. I checked my '/usr/local' directory, RBDL was not built into either '/include' or '/lib'.  And I can't seem to find where the RBDL library file is installed at otherwise.
Would someone point me to what's causing this issue?

ju6ge, 2021-11-09:
If you do a `sudo make install`, self build libraries will be copied to `/usr/local/lib` and headers to `/usr/local/include`.  The FindCmake file for RBDL will also look in these directories. So I assume you did not do the the `sudo make install`. 

If you don't want to install rbdl system wide you need to set the cmake option `CMAKE_INSTALL_PREFIX` to a installation folder where you have write access to. Executing `make install` will then copy the libraries and headers files to this folder. Then when you compile your code against RBDL or you compile any example you have to point the cmake option `CUSTOM_RBDL_PATH` to this installation folder. 

I hope this information helps ;) Kind regards
ju6ge

zcoco99, 2021-11-09:
Thanks so much for the pointer, that solved the problem! If you don't mind helping, I also haven't resolved the next checkpoint , `RBDL_LUAMODEL_LIBRARY` not found. 
I followed the instructions of using `sudo apt install lua5.1`, `sudo apt install liblua5.1-0-dev`.
Then in the rbdl build folder, I have turned on  `RBDL_BUILD_ADDON_LUAMODEL        ON`.
However, right now there's no `addons/luamodel` folder under `/usr/local/include/rbdl`.
Is there any step I'm missing here too?

zcoco99, 2021-11-09:
Ran the make command again after changing settings. All works now. Thank you very much!

closed 2021-11-09 by zcoco99