ORB-HD/rbdl-orb #82: Problem when install the muscle addon

Wayne-Z, 2021-12-08:
Hello,
In the process of 'Compiling Cython CXX source for rbdlmuscle-python', it was failed and reminded me that:
`LookupError: unknown encoding: default `
 `python/CMakeFiles/rbdlmuscle-python.dir/build.make:64: recipe for target 'python/rbdlmuscle-python.cxx' failed `
 `make[2]: *** [python/rbdlmuscle-python.cxx] Error 1 `
 `CMakeFiles/Makefile2:848: recipe for target 'python/CMakeFiles/rbdlmuscle-python.dir/all' failed `
 `make[1]: *** [python/CMakeFiles/rbdlmuscle-python.dir/all] Error 2 `
 `Makefile:151: recipe for target 'all' failed `

Thanks a lot for any help on this.

ju6ge, 2021-12-08:
Hey,

could you provide more information on the system you are using and what build options you are using? 

Generally i would advise that when you use cython you should install cython via pip and not the system package manager, because the version in pip is often more up to date and less buggy ;)

Kind regards,
ju6ge

Wayne-Z, 2021-12-08:
Thanks a lot. my Cython3 version is 0.26.1. I installed the cython via apt-get install but not pip, maybe I should try it as you said.

closed 2022-01-28 by ju6ge