ORB-HD/rbdl-orb #85: Add orientation constraint to InverseKinematics()

zcoco99, 2021-12-29:
I am using RBDL for inverse kinematics in Python. I have been successful in using `rbdl.InverseKinematics(model, q_init, body_id, body_point,target_pos, q_res)`. I'd now like to introduce orientation constraints to the function so that the robot approaches target in specified orientations. 
I see in the documentation that there's another way of using the IK function: `rbdl.InverseKinematics(model, q_init, IK_constraint_set, q_res)`, where the `IK_constraint_set` should include `body_id, body_point, target_pos, target_orientation`. 
How do I define the `IK_constraint_set` variable here?

I have tried the following:
```
q_init = np.array([0. , np.pi/2, 0. , -np.pi, np.pi/2 , 0. ])
q_res = np.zeros(n_joints) # Output of computed inverse kinematics

body_id = np.array([ee_link_id]) # id of end effector
body_point = np.array([[0.0, 0.0, 0.0]]) # body local coordinate
target_pos = np.array([[ 0.66141797, -0.04902463,  0.15271294]])
target_ori = np.array([0, 0, -np.pi/2])

IKconstraint = rbdl.InverseKinematicsConstraintSet()
cs = IKconstraint.AddFullConstraint(body_id, body_point, target_pos, target_ori)
rbdl.InverseKinematics(model, q_init, cs, q_res)
```

However this was not calling the IK function as I intended to, I get the following error: 
`TypeError: InverseKinematics() takes at least 6 positional arguments (4 given)`

zcoco99, 2021-12-29:
Update: Tried to add constraint using `rbdl.InverseKinematicsCS(model, q_init, IKconstraint, q_res)` instead. Using the same parameters as defined earlier, but the solution leads the end effector to somewhere near the origin, not the target position.

[image]
(Fig 1: Solution calculated with InverseKinematicsCS(). Green sphere shows the target position.)

[image]
(Fig2: Solution calculated with `rbdl.InverseKinematics(model, q_init, body_id, body_point,target_pos, q_res)`. End effector is aligned with the green target point.

ju6ge, 2021-12-29:
Hey zcoco99 as far as I can tell the `rdbl.InverseKinematicsCS' function is calling it's C counter part correctly. Appologies for the poor python documentation by the way. There are for sure some bugs to be found there.But I don't think that is the cause of your issue. 

Why are you creating the body_id as an vector?  A constraint always only applies to a single body of your multi-body robot. So AddFullConstrait only expects the integer body_id then (the target point on that body part relative to the origin of the body), (the target world position), (the target world orientation). Since constraint sets can be defined separately from the model there is no way for the function to check if the body_id actually exists in the model hence it can not tell you if the body id entered is valid. It is only in the Kinematics/Dynamics functions that the constraint sets and models are combined to perform the computations.

zcoco99, 2021-12-29:
In `rbdl.InverseKinematics(model, q_init, body_id, body_point, target_pos, q_res)`, `body_id` parameter was required as a np array, and I managed to get correct solutions using this function, so I believe the body id used was valid. The solution was not in the desired approach orientation, thus I'm trying the IK function with orientation constraint.
`def InverseKinematics (Model model, 
        np.ndarray[double, ndim=1, mode="c"] qinit, 
        np.ndarray[np.int64_t, ndim=1] body_ids, 
        np.ndarray[double, ndim=2, mode="c"] body_point_position, 
        np.ndarray[double, ndim=2, mode="c"] target_pos_position, 
        np.ndarray[double, ndim=1, mode="c"] qres, 
        double step_tol = 1.0e-12, 
        double lambda_ = 0.01, 
        int max_iter = 50):`

I just tried passing `body_id` as an interger for `rdbl.InverseKinematicsCS`, didn't seem to make a difference. The solution still lead the end effector to somewhere near the origin. 

I'm wondering how the target orientation should be represented? You mentioned it is in world orientation, should it be in xyz or zyz form? 
I tried xyz form, passing in `target_ori = np.array([ [1.0, 0., 0.] , [0., 1.0, 0.] , [0., 0., 1.0] ])` , I would expect the end effector to move to the target position, with orientation vertically upwards. But the solution had the end effector at the origin and parallel to floor level. I tried different values for the rotation matrix, but still couldn't get a solution that has the end effector orientation to upwards or downwards vertical.

ju6ge, 2021-12-30:
Hm that is interesting …

Regarding orientation, RBDL never explicitly defines which positions corresponds to which axis. All that is required for the math to work is for the coordinates to be right handed. How the coordinates map to axis is left to the developer of the model. This flexibility has its pittfalls, if you are not careful to be consistent. So without more information on what kind of a model you are using and with what tools you are further visualizing your results i can not really tell.

Does the orientation of the end effector at all change if you change the value of rotation matrix? If not it would be interesting to see if the same implementation in cpp has the same result. If not then that would point to something going wrong in the cython translation.