ORB-HD/rbdl-orb #86: URDFReadFromFile mistakes negative revolute joints as helical

samtkeen, 2022-02-01:
Here is a snippet from a robot URDF
```
  <joint name="joint_1" type="revolute">
    <origin rpy="0 0 0" xyz="0 0 1.180"/>
    <parent link="base_link"/>
    <child link="link1"/>
    <axis xyz="0 0 -1"/>
    <limit effort="0" lower="-2.9670" upper="2.9670" velocity="3.1415"/>
  </joint>
```
When accessing the joint type, ir returns "13" or "JointTypeHelical", even though the joint type is clearly revolute. by changing the urdf to this below
```
  <joint name="joint_1" type="revolute">
    <origin rpy="0 0 0" xyz="0 0 1.180"/>
    <parent link="base_link"/>
    <child link="link1"/>
    <axis xyz="0 0 1"/>
    <limit effort="0" lower="-2.9670" upper="2.9670" velocity="3.1415"/>
  </joint>
```
the joint type is recognised as revolute as it should be. I could not see any mention to the JointTypeHelical in the source code. Will this affect some of the functions, specifically the InverseKinetmatics?

ju6ge, 2022-02-02:
Hey @samtkeen 

Which version of RBDL are you using? And which compile options did you specify? 

Per default RBLD-ORB uses our [Urdf Parser](https://github.com/ORB-HD/URDF_Parser/blob/master/src/joint.cpp) to parse the joints. And which joint type gets set does not depend on the values of the axis, it only checks the type attribute and then saves the axis value to a vector. Also this code does not know about the joint type helical at all, so it is strange that you can get that type if you are using the default configuration.

Later the in the URDF Addon code which constructs the model, the joint type get's read to determine what Spacial Vector(s) need to be added to the model. Again there is no logic that changes the values of the axis so if you have a revolute joint with a negative axis, this negative value will also be set on the spacial vector so that the rotation should be in the other direction. That code also does not now about the helical type so if you get that you might end up with an joint with a zero spacial vector. That would be bad and probably result lead to bad results when using forward or inverse kinematics and dynamics.

samtkeen, 2022-02-02:
Hello,

I am using rbdl-orb 64 bit version from vcpackage as i couldnt get the addons to compile with the standard rbdl.

I did have a comb through on the github trying to find where it would be changing it to helical and couldnt find it so it makes sense the version i got from vcpackage is different but I'm not sure why it is that helical joints would be omitted from newer version? 

It is odd that changing th value of the axis can change the joint type though if the joint type if they are independant.

I will assume this isnt an issue on newer versions?

ju6ge, 2022-02-02:
Hm this does not make any sense … I wrote all of that code (URDF_Parser), as well as the rbd-orb port for vcpkg. 

The [RBDL-ORB VCPKG Buildfile](https://github.com/microsoft/vcpkg/blob/master/ports/rbdl-orb/portfile.cmake) also points to the newest commit on the parser code. And as far as I am aware the Helical type was never part of the codebase for the parser. RBDL has a build flag to use the ROS URDF Parser instead of our parser, but since you are using the VCPKG version that is not the case.

Could you try and compile the URDF_Parser library yourself and let it read your model file?