ORB-HD/rbdl-orb #87: Analytical Inverse of the Joint Space Inertia Matrix

Ipuch, 2022-02-03:
Hi,

We have implemented in biorbd the Analytical Inverse of the Joint Space Inertia Matrix proposed by [Justin Carpentier](https://hal.laas.fr/hal-01790934v2/document). We wanted to let you because it may be interesting for you to speed up some calculations. 

For now it only works with 1 Dof per joint, here is the code using rbdl through biorbd :
https://github.com/pyomeca/biorbd/blob/583e7c2ff67e02bd1387b902943a179c5f33a005/src/RigidBody/Joints.cpp#L802-L924

Regards,

ju6ge, 2022-02-03:
Hey @lpuch,

that is awesome :+1: … sadly my time to develop on rbdl is very limited at the moment :( But if it is desired to add this functionality to the main rbdl code I would be more than happy to merge it ;)

Here are my thoughts:
- so from the look of it this should be added to the RBDL [utils](https://github.com/ORB-HD/rbdl-orb/blob/master/include/rbdl/rbdl_utils.h) code 
- add a docsting to the function explaning what it does, and what constaints need to be considered
- add a test case to the rbdl test suite

With those things in place I would be happy to merge the corresponding PR and then bump the version of RBDL creating a new release with this functionality added :)

Kind regards,
ju6ge

Ipuch, 2022-02-03:
Thanks for the quick answer. For now, it works fine in Biorbd. I also have a limited amount of time to code. 

My question would rather be on if you see some potential use of this function to speed up some algorithms already implemented in Rdbl. If not I'm not sure if want to spent time on it.

I remain available if you need some help and question to retrieve the implementation in rbdl.

Regards,

Pierre