ORB-HD/rbdl-orb #91: Confused about function CalcBodySpatialJacobian

wangchangys, 2022-02-22:
**I built a lua model and test it with python, its a 3DOF robot arm:**

`
print("Lua: Constructing robot_3DOF")

inertiaMatrix = {
    {2.0,0.     ,0.    },
    {0.    ,2.0,0.    },
    {0.    ,0.     ,2.0}
}
print("Lua: Body mass and geometry properties")
bodies = {
    base  = {   mass    = 1.,
                com     = {0.,0.,0.},
                inertia = inertiaMatrix},
    link1  = {   mass    = 1.,
                com     = {0.,0.0,0.},
                inertia = inertiaMatrix},
    link2  = {   mass    = 1.,
                com     = {0.,0.0,0.},
                inertia = inertiaMatrix},
    link3  = {   mass    = 1.,
                com     = {0.,0.0,0.},
                inertia = inertiaMatrix},
}

print("Lua: Making the model")
model = {
    gravity = {0., 0, -9.81},

    frames = {    
        {
            name    = "K0",
            parent  = "ROOT",
            body    = bodies.base,
            joint   = {},
            joint_frame = {
                r = {0.1, 0.2, 0.3},
            }
        },
        {
            name    = "K1",
            parent  = "K0",
            body    = bodies.link1,
            joint   = {{0.,0.,1.,0.,0.,0.}},
            joint_frame = {
                r = {0,0, 0},
            }
        },
        {
            name    = "K2",
            parent  = "K1",
            body    = bodies.link2,
            joint   = {{1,0,0,0,0,0}},
            joint_frame = {
                r = {0,0, 5},
            }
        },
        {
            name    = "K3",
            parent  = "K2",
            body    = bodies.link3,
            joint   = {{0,0,1,0,0,0}},
            joint_frame = {
                r = {0,5, 0},
                --the element in E is colum
                E = {
                 {0., 1., 0.},
                  {0., 0., 1.},
                  {1., 0., 0.}
                }
            }
        },
        {
            name    = "tool",
            parent  = "K3",
            joint_frame = {
                r = {6.,0., 0.},
            }
        },
    }

}

return model

`

**I want get its jacobian matrix, the python code is:**

> import numpy as np
> import rbdl
> model = rbdl.loadModel('./model/robot_3DOF.lua')
> q = np.zeros(model.q_size)
> qdot = np.zeros(model.qdot_size)
> qddot = np.zeros(model.qdot_size)
> tau = np.zeros(model.qdot_size)
> tool_p1_tool = np.array([0., 0., 0.])
> tool_p1_0 = rbdl.CalcBodyToBaseCoordinates(model, q, model.GetBodyId('tool'), tool_p1_tool, True)
> print('tool body point p1 in {0} system:\n', tool_p1_0)
> G = np.zeros([model.qdot_size, 6])
> rbdl.CalcBodySpatialJacobian(model, q, model.GetBodyId('tool'), tool_p1_tool, G, True)
> print('spatial jacobian of tool body:\n', G)

**The question is: if the SpatialJacobian is the mapping between " the spatial velocity of a body at the origin of coordinate system of body" and the input dotq, why the python function "CalcBodySpatialJacobian" need "tool_p1_tool" ? And I have checked the document, it says:**

RBDL_DLLAPI void CalcBodySpatialJacobian | ( | Model & | model,
-- | -- | -- | --
  |   | const Math::VectorNd & | Q,
  |   | unsigned int | body_id,
  |   | Math::MatrixNd & | G,
  |   | bool | update_kinematics = true
  | ) |  

**It acturally dont need the point input, but why the python code need ?**

ju6ge, 2022-02-22:
This seems like an inconsistency between the python and the c++ code base. I don't think that the python code needs that input. The python code mostly only converts between the python and cpp types and then calls the cpp implementation via the python bindings generated by cython.

The python code has not received as much attention as the cpp code. I am sure there are more such cases to be found.
Since RBDL is the work of a research group updates only happen when somebody takes the time to fix the issues they encounter. 

This seems like a fairly straightforward fix though, so i would encourage you to take a look at the python implementation and try to fix it yourself. Sadly my time is very limited at the moment. Apart from answering issues and merging pull request I can not do active development for the time beeing :see_no_evil:

wangchangys, 2022-02-22:
@ju6ge **I also test the "CalcPointJacobian6D" function in C++:**
`
#include <string>
#include <iostream>
#include <stdio.h> 
#include <rbdl/rbdl.h>
#include "csvtools.h"

#include <boost/numeric/odeint/stepper/runge_kutta_cash_karp54.hpp>
#include <boost/numeric/odeint/stepper/controlled_runge_kutta.hpp>
#include <boost/numeric/odeint/integrate/integrate_adaptive.hpp>
#include <boost/numeric/odeint/stepper/generation/make_controlled.hpp>
//using namespace std;
using namespace boost::numeric::odeint;

#ifndef RBDL_BUILD_ADDON_LUAMODEL
    #error "Error: RBDL addon LuaModel not enabled."
#endif

#include <rbdl/addons/luamodel/luamodel.h>

using namespace RigidBodyDynamics;
using namespace RigidBodyDynamics::Math;

int main (int argc, char* argv[]) {
    rbdl_check_api_version (RBDL_API_VERSION);

    VectorNd q, qd;
    MatrixNd G,G2;
    Vector3d p1;

    Model* model  = NULL;
    model         = new Model();
    int i=0;

    if (!Addons::LuaModelReadFromFile ("./../model/robot_3DOF.lua", 
                                       model, false)             ){        
        std::cerr     << "Error loading model ./model/pendulum.lua" 
                    << std::endl;
        abort();
    }
    

    q       = VectorNd::Zero (model->dof_count);
    qd      = VectorNd::Zero (model->dof_count);
    G      = MatrixNd::Zero (6,3);
    G2     = MatrixNd::Zero (6,3);
    p1={0,0,0};

    printf("DoF: %i\n",model->dof_count);
    CalcBodySpatialJacobian(*model,q,model->GetBodyId("tool"),G,true);

    CalcPointJacobian6D(*model,q,model->GetBodyId("tool"),p1,G2,true);

    printf("body tool spatial jacobian is\n");
    for(i=0;i<6;i++)
    {
        printf("%f\t%f\t%f\n",G(i,0),G(i,1),G(i,2));
    }

    printf("point jacobian is\n");
    for(i=0;i<6;i++)
    {
        printf("%f\t%f\t%f\n",G(i,0),G(i,1),G2(i,2));
    }

    delete model;

    return 0;
        
}

`
**The program result is:**

Lua: Constructing robot_3DOF
Lua: Body mass and geometry properties
Lua: Making the model
DoF: 3
body tool spatial jacobian is
0.000000	0.000000	0.000000
1.000000	0.000000	0.000000
0.000000	1.000000	1.000000
0.000000	0.000000	0.000000
0.000000	11.000000	6.000000
-11.000000	0.000000	0.000000
point jacobian is
0.000000	0.000000	1.000000
1.000000	0.000000	0.000000
0.000000	1.000000	0.000000
0.000000	0.000000	0.000000
0.000000	11.000000	0.000000
-11.000000	0.000000	6.000000

The point jacobian seem not right. if the point jacobian is "gives a 6-D vector that has the angular velocity as the first three entries and the linear velocity as the last three entries.", as the document said, the point jacobian result seems to be:
0,1,1
0,0,0
1,0,0
-11,0,0
0,0,0
0,11,0

ju6ge, 2022-02-23:
If the function signature in python is different, this does not surprise me much. Cython would then incorrectly map the function call from python to c++. The results of the c++ implementation are the correct ones, since the c++ implementation is tested thoroughly. 

Take a look at the [cython function definitions](https://github.com/ORB-HD/rbdl-orb/blob/master/python/crbdl.pxd) as well es the [python call](https://github.com/ORB-HD/rbdl-orb/blob/master/python/rbdl-wrapper.pyx) to CalcBodySpacialJacobian. Updating these functions to have the correct parameters should resolve the issue since cython would then correctly build the python bindings.