ORB-HD/rbdl-orb #92: Segmentation fault when use CalcAssemblyQ for a four-bar mechanism

wangchangys, 2022-02-24:
_Hello, I buit a closed loop  four bar mechanism with three revolut joint and the last joint is connected with root by constraints. The lua script is :_
print("Lua: start Constructing four link")
l1=1
l2=2
l3=3
p_x0=4
p_y0=0.5
inertiaMatrix = {
  {2.0,0.     ,0.    },
  {0.    ,2.0,0.    },
  {0.    ,0.     ,2.0}
}
bodies = {
  link1  = {   mass    = 1.,
              com     = {0.,0.0,0.},
              inertia = inertiaMatrix},
  link2  = {   mass    = 1.,
              com     = {0.,0.0,0.},
              inertia = inertiaMatrix},
  link3  = {   mass    = 1.,
              com     = {0.,0.0,0.},
              inertia = inertiaMatrix},
}
return {
gravity = {0., -9.81, 0.},
configuration = {
  axis_front  = { 1,  0, 0,},
  axis_right  = { 0, -1, 0,},
  axis_up     = { 0,  0, 1,},
},
constraint_sets = {
  PinJoint = {
    {constraint_type = "loop", 
    name = "x",
    predecessor_body = "ROOT",
    successor_body = "K4",
    predecessor_transform = {
      E = {
        {1, 0, 0},
        {0, 1, 0},
        {0, 0, 1},
      },
      r = {p_x0, p_y0,  0.000},
    },
    successor_transform = {
      E = {
        {1, 0, 0},
        {0, 1, 0},
        {0, 0, 1},
      },
      r = {0.0, l3, 0.0,},
    },
    axis = {0, 0, 0, 1, 0, 0},
    stabilization_coefficient = 1.0,
    baumgarte_enabled = false,
   },
   {constraint_type = "loop", 
    name = "y",
    predecessor_body = "ROOT",
    successor_body = "K4",
    predecessor_transform = {
      E = {
        {1, 0, 0},
        {0, 1, 0},
        {0, 0, 1},
      },
      r = {p_x0, p_y0,  0.000},
    },
    successor_transform = {
      E = {
        {1, 0, 0},
        {0, 1, 0},
        {0, 0, 1},
      },
      r = {0.0, l3, 0.0,},
    },
    axis = {0, 0, 0, 0, 1, 0},
    stabilization_coefficient = 1.0,
    baumgarte_enabled = false,
   },
   {constraint_type = "loop", 
    name = "z",
    predecessor_body = "ROOT",
    successor_body = "K4",
    predecessor_transform = {
      E = {
        {1, 0, 0},
        {0, 1, 0},
        {0, 0, 1},
      },
      r = {p_x0, p_y0,  0.000},
    },
    successor_transform = {
      E = {
        {1, 0, 0},
        {0, 1, 0},
        {0, 0, 1},
      },
      r = {0.0, l3, 0.0,},
    },
    axis = {0, 0, 0, 0, 0, 1},
    stabilization_coefficient = 1.0,
    baumgarte_enabled = false,
   },
  },
},
frames = {    
        {
            name    = "K1",
            parent  = "ROOT",
            body    = bodies.link1,
            joint   = {{0.,0.,1.,0.,0.,0.}},
            joint_frame = {
                r = {0.0, 0.0, 0.0},
            }
        },
        {
            name    = "K2",
            parent  = "K1",
            body    = bodies.link2,
            joint   = {{0.,0.,1.,0.,0.,0.}},
            joint_frame = {
                r = {0.,l1, 0.},
            }
        },
        {
            name    = "K3",
            parent  = "K2",
            body    = bodies.link2,
            joint   = 
            {{0.,0.,1.,0.,0.,0.},},
            joint_frame = {
                r = {0.,l2, 0.},
            }
        },
    },
}

_The C++ code is:_

```
#include <string>
#include <iostream>
#include <iomanip>
#include <stdio.h> 
#include <rbdl/rbdl.h>
#include "csvtools.h"

#include <boost/numeric/odeint/stepper/runge_kutta_cash_karp54.hpp>
#include <boost/numeric/odeint/stepper/controlled_runge_kutta.hpp>
#include <boost/numeric/odeint/integrate/integrate_adaptive.hpp>
#include <boost/numeric/odeint/stepper/generation/make_controlled.hpp>
//using namespace std;
using namespace boost::numeric::odeint;

#ifndef RBDL_BUILD_ADDON_LUAMODEL
    #error "Error: RBDL addon LuaModel not enabled."
#endif

#include <rbdl/addons/luamodel/luamodel.h>
#include <rbdl/addons/luamodel/luatables.h>

using namespace std;
using namespace RigidBodyDynamics;
using namespace RigidBodyDynamics::Math;

typedef std::vector< double > state_type;

typedef runge_kutta_cash_karp54< state_type > error_stepper_type;
typedef controlled_runge_kutta< error_stepper_type > controlled_stepper_type;

/* Problem Constants */
int main (int argc, char* argv[]) {
    rbdl_check_api_version (RBDL_API_VERSION);

    //problem specific constants
    int     nPts    = 100;
    double  t0      = 0;
    double  t1      = 3;

    size_t i=0;

    VectorNd q_init,q_res,q_w,qd, qdd, tau;

    RigidBodyDynamics::Model model;
    string modelFile = "../model/four_link.lua";

    std::vector<std::string> constraintSetNames = 
        Addons::LuaModelGetConstraintSetNames(modelFile.c_str());
    std::vector<RigidBodyDynamics::ConstraintSet> constraintSets;

    constraintSets.resize(constraintSetNames.size());

    

    if (! Addons::LuaModelReadFromFileWithConstraints(
                                      modelFile.c_str(),
                                      &model,
                                      constraintSets,
                                      constraintSetNames,
                                      false)            ){
        std::cerr     << "Error loading model" << std::endl;
        abort();
    }

    printf("DoF: %i\n",model.dof_count);

    q_init       = VectorNd::Zero (model.dof_count);
    q_res   = VectorNd::Zero (model.dof_count);
    q_w     = VectorNd::Zero (model.dof_count);
    qd      = VectorNd::Zero (model.dof_count);
    qdd     = VectorNd::Zero (model.dof_count);
    tau     = VectorNd::Zero (model.dof_count);

    q_w[0]=1;
    q_w[1]=0.01;
    q_w[2]=0.01;

    q_init[0]=0;
    q_init[1]=-70/180*3.14;
    q_init[2]=-60/180*3.14;

    std::cout << constraintSets[0].name[0] << std::endl;
    std::cout << constraintSets[0].name[1] << std::endl;
    std::cout << constraintSets[0].name[2] << std::endl;

    CalcAssemblyQ(model,q_init,constraintSets[0],q_res,q_w,0.0001,100);

    return 0;
        
}
```

_function CalcAssemblyQ give an error: Segmentation fault, and i dont know why. Could anyone give some help ?_

ju6ge, 2022-02-24:
When encountering Segmentations Faults it is useful to run the program using gdb. It can give you a stack trace to exactly tell you at which point the code fails :wink: 

Steps:
- `gdb ./binary'
Then in gdb command line type:
- `r` for run ;)
- `backtrace` to see the stack trace

wangchangys, 2022-02-25:
I have tried the debug, but the problem cannot be solved, and the source codes seem hard for me. I am not sure whether the four-bar close loop mechanism can be build like this: with three revolute joint and 3 translational constrains of the end point of the third link and the base(i also tried to just restraint the x and y translational movement only , but the error is throwed out too). Or the problem may be that i called CalcAssemblyQ function in a wrong way?

wangchangys, 2022-02-26:
I finally find the reason. It is a stupied mistake:  the constraint_sets definition in lua file "successor_body = "K4", it should be K3.

closed 2022-02-26 by wangchangys

ju6ge, 2022-02-26:
Great to hear that :+1: … It always the little things :laughing: