ORB-HD/rbdl-orb #93: Rbdl installation problem

tianbaoloin, 2022-03-16:
I am a newbie and not very familiar with rbdl and cmake，Thank you for your hard work，When I installed step 8
_Build RBDL using CMake (http://www.cmake.org/). To compile the library in a separate directory in Release mode use:
  mkdir /rbdl-build
  cd rbdl-build/
  cmake -D CMAKE_BUILD_TYPE=Release ../rbdl
  make_
I want to test the code，But it keeps prompting that the command does not exist，Why does the command not exist?（RBDL_BUILD_TESTS    ON）
_RBDL_BUILD_TESTS                 ON
RUN_AUTOMATIC_TESTS              ON_
When I execute this command to bind python, it also prompts that it does not exist，
_Build and install RBDL with the
RBDL_BUILD_PYTHON_WRAPPER : ON_
Can you give me some installation advice, thank you very much for your work, looking forward to your reply

ju6ge, 2022-03-16:
cmake only generates the build files for rbdl, to actually build the software you also have to run `make` in your build folder!

tianbaoloin, 2022-03-16:
> cmake only generates the build files for rbdl, to actually build the software you also have to run `make` in your build folder!
I just ran the command sudo make install, then rebooted and it works for now, thanks for your prompt reply.

ju6ge, 2022-03-16:
No problem ;)

closed 2022-03-16 by ju6ge