ORB-HD/rbdl-orb #96: Python Binding Error

ggorgeousx, 2022-04-22:
Good afternoon! I have a problem when installing RBDL with the Python binding and I hope you can help.
【Information】
Ubuntu 20.04
Python 3.9.7
Cython 0.29.24

【Errors】
[image]
[image]
[image]

and at the end it tells me:
```
make[2]: *** [python/CMakeFiles/rbdl-python.dir/build.make:68: python/CMakeFiles/rbdl-python.dir/rbdl-python.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:202: python/CMakeFiles/rbdl-python.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

1. For error1, maybe there is also a Cython problem?
But I used `pip3 install cython` and it didn't work.

2. For error2, I have set the path 
`cmake -DPYTHON_INCLUDE_DIR=/home/guojue/anaconda3/include/python3.9 -DPYTHON_LIBRARY=/home/guojue/anaconda3/lib` 
Do I need to set the path elsewhere?

Hope to get your help! Thank you very much : )

ju6ge, 2022-04-22:
If you are using Simple Math you are not using rbdl-orb, since Simple Math was removed from this version due to incompatibilities with some addons. Why wouldn't you just use eigen3? It is more performant anyway.

ggorgeousx, 2022-04-23:
Thank you for your answer!
I'm trying to reproduce a paper that requires binding RBDL to python, I'm not sure if it's possible using eigen3

ju6ge, 2022-04-23:
Eigen3 is a well tested and widely used Mathematics library. The only reason for the existence of Simple Math as far as I could tell was faster compile times for development. But nobody is using it to produce scientific results! The reason we removed simple math was exactly this, confusion and incompatibilities, modern machines compile fast enough for development with Eigen and Eigen is far more performant and reliable. Because of this things like the python addon only support eigen, because it is not worth the effort to maintain compatibility to a math backend that is not used in practice. So there is no way in hell the paper you are trying to reproduce uses Simple Math.

closed 2022-05-05 by ju6ge