ORB-HD/rbdl-orb #98: RBDL Coordinate system

Adrianrr94, 2022-05-05:
Hello,

I have been reading the rbdl-toolkit wiki and it seems like the coordinate system of the rbdl-toolkit is {x, y, z} = {right, up, front}. I understand that the coordinate system of the rbdl library is the same. Am I right?

Thank you.

Regards.

ju6ge, 2022-05-05:
Well for sake of simplicity I would suggest you stick with the corrdinate system of the toolkit (actually it is the opengl coordinate system :wink: ). RBDL itself does not care about what each axis represents, it only requires the system to be right handed.

Adrianrr94, 2022-05-05:
Thanks for the quick reply.

I have one more question somehow related to the frame system. Why do you transpose the rotation to calculate the position of a body in the function CalcBodyToBaseCoordinates and then return the rotation in the CalcBodyWorldOrientation?

Thanks in advance.

ju6ge, 2022-05-05:
That is a much more technical question, the answer probably lies in the mathematics involved. I can't give you good enough answer about this right now, mostly because I am not the one how originally wrote this code. I mostly do code maintainer ship and at the moment I do not have time to do a deep dive in mathematics just to answer the question. Sry … but I you are interested in understanding this, do your own research, I am sure you can figure it out ;) And if you do, you can post the answer here, so that more people can benefit.

Adrianrr94, 2022-05-05:
Okay, I understand, I am sorry but I don't have much time either to do so and there are some operations when adding a body to the model that affects these functions and I still don't understand. 
I will close the issue.

Thanks for the quick responses.

closed 2022-05-05 by Adrianrr94