ORB-HD/rbdl-orb #100: Difference of 180 def in pitch and yaw rotation between rbdl and gazebo

Adrianrr94, 2022-05-06:
Greetings,

I am creating a model by urdf file and getting the orientation using the function CalcBodyWorldOrientation and the position with CalcBodyToBaseCoordinates. 
At the same time, I am simulating the same urdf arm in gazebo and getting the pose of the end effector, it seems like there is a difference of 180 deg in pitch and yaw while the roll and the position are identical.

Any idea what is happening?

Thanks.

ju6ge, 2022-05-06:
Well that seems like it is a redundant solution, meaning a solution that describes exactly the same position and orientation, since when you pitch a symmetric object by 180 degrees and the yaw by 180, the overall orientation is the same as before. These redundant results can arise in different situations, and are typically the result of not enough constraints on the solution.

Adrianrr94, 2022-05-06:
But it is not a symmetric object, it is robotic arm link mesh and as far as I know constraints has nothing to do with functions CalcBodyWorldOrientation and CalcBodyToBaseCoordinates, only the relation between links defined in the urdf file.

ju6ge, 2022-05-06:
The math does not care what shape the object is, in fact it does not even know the shape, because rbdl does not use that information in any way. So it depends on how the model is constructed and what you are actually doing. In general there are multiple valid values for any given position.

closed 2022-05-06 by ju6ge